kisslorand/BTT-TFT-FW #1: PID tuning stuck screen

patriko123, 2023-08-13:
Hello when I do PID autotune I remain stuck in screen Busy processing please wait. After dome time is on screen pid tuning complete but i remain stuck and then after some Timeout reached! Busy processing please wait. Should this work like that ?

kisslorand, 2023-08-13:
Please provide more details. For what did you do the PID tuning?

patriko123, 2023-08-13:
For hotend and bed

kisslorand, 2023-08-13:
I just tested PID autotune and cannot reproduce the mentioned issue.
Try, if you can, to see if you have the same issue when starting PID tune from terminal or any other guest connected to the motherboard (Octoprint, Pronterface, etc). If yes you should look at your motherboard configs.

kisslorand, 2023-10-08:
Closing it due to inactivity.

closed 2023-10-08 by kisslorand