kisslorand/BTT-TFT-FW #4: Wrong bed leveling points

Millissimo, 2023-09-27:
Hi,
I have installed the tft firmware on an Artillery Sidewinder X2 with the GD32F305 microcontroller. After the installation, I noticed that when proceeding with the manual bed leveling the points were in the wrong position. To be more clear I made a quick picture, if you imagine that the square is the bed seen from above the blue dots are the points at which the leveling is done with the original firmware, while the red dots are the points at which the leveling is performed with the new firmware. 

[image]

I have tried to level the bed with the old firmware and the points return to the original position. I have also noticed that when homing with the original firmware when the procedure is finished the extruder moves slightly to the right centering the nozzle to the center of the bed, while this does not happen with the BTT firmware, can these two things be related? Is there any way to fix this?
Thank you

kisslorand, 2023-09-28:
Do you use "home offset"? Do you use the latest bugfix version of Marlin?

Millissimo, 2023-09-28:
I have already checked in the home offset menu and X,Y and Z are all set to zero. 
Yes, I am using the latest release of Marlin. I even tried to use the one present on the Artillery site and it makes no difference, while with the original TFT firmware the problem disappear.

kisslorand, 2023-09-28:
Do you use the latest __release__ or the latest __bugfix__ version?

Millissimo, 2023-09-28:
Sorry, I mean latest bugfix.

kisslorand, 2023-09-28:
Can you please provide the response of the printer to the "M115" command?
I am interested in the response that looks like:
"area:{full:{min:{x:0.000,y:0.000,z:0.000},max:{x:305.000,y:305.000,z:390.000}},work:{min:{x:0.000,y:0.000,z:0.000},max:{x:305.000,y:305.000,z:390.000}}}"

Millissimo, 2023-09-28:
It took me a moment because I did not have M115_GEOMETRY_REPORT enabled so I had to reflash the printer. However, this is the response:

area:{full:{min:{x:-2.00,y:-5.00,z:0.00},max:{x:300.00,y:300.00,z:400.00}},work:{min:{x:0.00,y:0.00,z:0.00},max:{x:300.00,y:300.00,z:400.00}}}

Where I can find the TFT variant? 

Thank you so much for your help, I appreciate it.

kisslorand, 2023-09-28:
> It took me a moment because I did not have M115_GEOMETRY_REPORT enabled so I had to reflash the printer. 

Please read the readme,  M115_GEOMETRY_REPORT is a must have option to be enabled (among many others)
<br>

> area:{full:{min:{x:-2.00,y:-5.00,z:0.00},max:{x:300.00,y:300.00,z:400.00}},

Is your bed 302mm X 305mm?
<br>

> Where I can find the TFT variant?

You must have known your variant of the TFT since you flashed it. What was the filename of the FW you flashed
<br>

Is your bed size defined correctly in the "config.ini" of the TFT FW?

Millissimo, 2023-09-29:
> Please read the readme, M115_GEOMETRY_REPORT is a must have option to be enabled (among many others)

I reflashed the printer making sure that all the needed options were enabled, I had previously missed one more other than M115_GEOMETRY_REPORT and I think that was the problem.

> Is your bed 302mm X 305mm?

Technically the bed of the sidewinder is 310x310mm, however, in the marlin configuration is set to 300mm. It is not very clear to me why X_MIN_POS and Y_MIN_POS are set differently from zero (I use the marlin example configuration for Sidewinder X2 as a starting point), nevertheless if I put zero the leveling points are slightly off, while with that offset are perfect.

> You must have known your variant of the TFT since you flashed it. What was the filename of the FW you flashed

I have flashed: MKS_GD_TFT28_V4_0.27.x Sound_Fixed

> Is your bed size defined correctly in the "config.ini" of the TFT FW?

I set it to 300x300mm as in marlin.

With the new marlin firmware, the leveling points are now perfect, thank you very much for your help.

closed 2023-10-08 by kisslorand