kisslorand/BTT-TFT-FW #5: TFT in listening mode!

discip, 2023-10-07:
I decided to give your prebuilds a try and flashed it to my Artillery Sidewinder X1 with the default MKS28 V4 and a SKR3EZ (723 MCU) board, which I have to try as it was suffering from hesitation / blobs total still stand.

As I have another printer equipped with also a SKR3EZ (743 MCU) and the TFT43 V3.0 suffering from the same suffering as well I did flash it with the respective build provided by you, but unfortunately it is inoperable. Displaying TFT in listening mode!

Furthermore I noticed that in **`Feature > Move speed (X Y Z)`** the default value was **`Englisch`** instead of **`slow`**, **`normal`** or **`fast`**.

Please help.

Thanks in advance
kind regards

kisslorand, 2023-10-07:
Please make an empty "reset.txt" on the SD Card  and reboot your printer with the SD card inserted. After the printer has been reset, copy the "config.ini" to the SD card and reboot your printer again with the SD card inserted. Please note that the "config.ini" file should be edited to suit your needs and to suit the setup of the printer.

discip, 2023-10-07:
First of all thank you for the fast answer.

I did as I was instructed, but unfortunately this did only partially solve my problem, as now there is the prompt **`Busy processing, please wait ...`** as soon as any kind of removable media (SD, USB) is inserted into the TFT. And even after removing it this message stays until the reset button is pressed!

discip, 2023-10-07:
And since we are on it, I hope you won't force me to open a separate issue, on my Artillery the **`Z Steppers Auto-Alignment`** does not work as reliably as it did prior to updating with your build.

FYI:
In the past, when I tried this feature the first time, I had a comparable issue, but managed to solve it by switching the places of the 2 Z-Stepper connectors.

kisslorand, 2023-10-07:
> First of all thank you for the fast answer.
> 
> I did as I was instructed, but unfortunately this did only partially solve my problem, as now there is the prompt **`Busy processing, please wait ...`** as soon as any kind of removable media (SD, USB) is inserted into the TFT. And even after removing it this message stays until the reset button is pressed!
> 
> If the reset button is pressed after the SD or USB is inserted or before bootup there is no prompt.
> 
> Seems to be triggered by plugging / unplugging removable media.

It's a known inconvenience and very soon there will be a big update that will solve this also.

discip, 2023-10-07:
Please forgive me, the "forcing" was intended as a joke.
I did not think this through before posting.

I really appreciate your work and did not mean to be rude or unrespectful.

About the topic:
The printer works, but I still have to test if the random pausing is solved.

Other than that, I did notice one more thing:
As I'm using thumbnails I noticed that they disappear if you go to the next page and only come back after the drive was excited and re-entered.

Thank you
Regards

kisslorand, 2023-10-08:
Updates are up, check them out!

discip, 2023-10-08:
## Thank you, for the fast reaction! 👍🏻

Have to actually print something to confirm this is working!
Will report as soon as I have results!

> If you do not like it tell it only to me, if you like it, tell it to everybody! :)

Will do! 😀

Closing, since the original issue is solved.

closed 2023-10-08 by discip