kisslorand/BTT-TFT-FW #7: Thumbnails disappearing when going through files.

discip, 2023-10-08:
As I'm using thumbnails I noticed that they disappear if you go to the next page and only come back after the drive was excited and re-entered.

kisslorand, 2023-10-08:
It's something I have to check on the original firmware. Will come back with the results, I am sure it can be solved.

discip, 2023-10-08:
I remember having a similar issue with the original firmware.

If I found the right spot than this should be the fix:

https://github.com/digant73/BIGTREETECH-TouchScreenFirmware/commit/82d754bee8fc524d3b4db8579fd540b6b410ede7

kisslorand, 2023-10-08:
The link you gave me is about the file extension, not about thumbnails.

discip, 2023-10-08:
Yes, but it is somehow related to this issue as discussed here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2300#issuecomment-1003675156

kisslorand, 2023-10-08:
OK. I will look into it.

kisslorand, 2023-10-09:
The issue has been found, working on a fix for it.

kisslorand, 2023-10-09:
It has been fixed, repository is updated with the fixes. Thank you!

discip, 2023-10-09:
## Clearly I'm the one having to thank you! 😃👍

Will test as soon as I'm home again.

discip, 2023-10-09:
## Unfortunately there was a print running up until now, which on the other hand proved the hesitation problem to be solved! 👍🏻 🎉

# After flashing the latest firmware the previously described ISSUES are GONE!
# Thank you so much!

closed 2023-10-09 by discip

discip, 2023-10-09:
May I ask if it makes sense to improve this browsing behavior even further without affecting performance?

I was thinking about storing the _'**thumbnails-pages**'_ after loading so that they don't have to be loaded individually each time the files are browsed.

kisslorand, 2023-10-10:
There is no place to store them, they are loaded directly into the RAM of the display driver as the file is accessed.

discip, 2023-10-10:
Alright.
Thanks for the clarification! 👍