kisslorand/BTT-TFT-FW #8: config.ini not compatible with TFT43

discip, 2023-10-08:
I tried both:

previous version 
**`"Machine Size / Build Area"`**

as well as

latest version
**`"Motion limits"` and `"Bed size"`**-version

Both resulting in:

    Invalid Keyword:

    size_min: ...

and

    Invalid Keyword:

    size_max: ...

And one additional little issue related to TFT43 (maybe other too but definitely **NOT MKS28**):

The text on the info-screen about the System, Machine, Board ... is shifted down so that the lowest lines are not legible. 😅

kisslorand, 2023-10-08:
There should be no such keyword as "size_min" and "size_max" in the "config.ini". I updated "config.ini", please check it out.

discip, 2023-10-08:
> There should be no such keyword as "size_min" and "size_max"

My bad I missed exactly that part. 😅
Its working now! 👍🏻

> I will look into it.

Thanks in advance!

closed 2023-10-08 by discip

kisslorand, 2023-10-09:
> The text on the info-screen about the System, Machine, Board ... is shifted down so that the lowest lines are not legible. 😅

Issue has been found and fixed for the next release.

discip, 2023-10-09:
## Great! 👍🏻 Your responsiveness and care are highly commendable!

kisslorand, 2023-10-09:
The repository is updated with the fixes.

discip, 2023-10-09:
I'm speechless!
Wow!

Will test as soon as I get home.