kisslorand/BTT-TFT-FW #10: Print Completes but TFT doesn't switch "Stop" button to "back" button and timer continues to run.

buryboi, 2023-12-17:
On TFT28 4.0.27.bin Came from digant's TFT.  I have to manually Stop the print which "cancels" it.  Do I need a certain gcode at the end of prints?

kisslorand, 2023-12-17:
Check your gcode file please. Is the last line an empty line? If not, than add an empty line to the end of the gcode file.

buryboi, 2023-12-17:
[Cube_PETG_42s.zip](https://github.com/kisslorand/BTT-TFT-FW/files/13698037/Cube_PETG_42s.zip)

It appears to have 3 empty lines at the end. (Attached as a zip).  Thanks, I reflashed Digant and confirmed the print completes as expected.

kisslorand, 2023-12-17:
> It appears to have 3 empty lines at the end.

Make it only one.

buryboi, 2023-12-18:
That seems to have solved it!  Is there a slicer script to cut the extra "returns" from the end of the g-code? (I have been running Orca Slicer recently).  thank you!

kisslorand, 2023-12-22:
> Is there a slicer script to cut the extra "returns" from the end of the g-code?

Not that I know of.
I will fix it in the TFT FW.

kisslorand, 2023-12-31:
The issue has been fixed.

closed 2023-12-31 by kisslorand