kisslorand/BTT-TFT-FW #14: MKS TFT28 Issue with Fonts

SpankyDevil, 2024-05-13:
MKS TFT28
Firmware MKS_TFT28_V3.0.27.x.bin
config.ini (Only changed Serial Port 1 baudrate to 250000)
TFT28 THEME_Unified Menu Material theme

When updating Fonts SD:/TFT28/font/word_unicode.fon gets to 25% and then changes to unreadable characters.
Once the update is complete the Dashboard is unreadable.

[image]
[image]
[image]
[image]
[image]
[image]

kisslorand, 2024-05-14:
Please try this:

- create an empty "reset.txt" file on your SD Card
- put the SD Card into the TFT's SD Card slot
- start/reboot the TFT
- restart the firmware update procedure including the addons but do not include any language file ("language_xx.ini") this time 

Let me know of the outcome of the above.

SpankyDevil, 2024-05-14:
1. Did a full format on an 8GB SD Card.
2. Created reset.txt
3. put the SD Card into the TFT's SD Card slot
4. start/reboot the TFT
5. Completed Touch Calibration
[image]

6. Deleted reset.DONE
7.Placed the following files on the SD Card
**Firmware MKS_TFT28_V3.0.27.x.bin** - renamed to MKSTFT28.bin
**config.ini** (Only changed Serial Port 1 baudrate to 250000)
**TFT28** - folder from THEME_Unified Menu Material theme

Results still not in English.

Here is the About screen and the change language screen.
[image]
[image]

kisslorand, 2024-05-15:
Can you please post a picture of the back  of your TFT board (where the MCU is)?

SpankyDevil, 2024-05-15:
[image]

kisslorand, 2024-05-15:
It seems you have a V2 TFT board which is not compatible with this firmware.

kisslorand, 2024-05-15:
It's kind of hard to do it without having the TFT.

I recommend to buy a compatible TFT, it should be really cheap nowadays.

SpankyDevil, 2024-05-15:
Fully understand.
I am happy to work with you and accept any risk of bricking the device if this is an option?
I feel this shouldn't need too many changes since it is working just an issue with the language or fonts.

I don't know if it makes any difference but the Original MKS TFT28_32_v3.0.6 firmware works correctly but doesn't have all the cool features of this firmware.

I am a Senior System Engineer so have lots of experience with flashing firmware, Serial connections, and some programming knowledge but not to this level required.

I find it strange that it is all in English until it gets to 25% flashing "word_unicode.fon" then it changes to something that looks like Chinese but I do not think it is Chinese.

SpankyDevil, 2024-05-16:
I deleted the word_unicode.fon file from within the TFT28\Font\ Folder and created a blank file with the same name.
Reflashed and now everything is working in English.

[image]
[image]
[image]

kisslorand, 2024-05-17:
Just as I thought, memory corruption/limitation is the culprit of your issues.
As you can see 193.65% SPIFlash is used. This FW's addons just doesn't fit into the memory of your TFT. You have only 4MB of SPI flash, 8MB is needed.

SpankyDevil, 2024-05-17:
How is that even possible to store nearly 8gb on a 4gb.
Im surprised its running so well.

I guess thats the difference between the MKS TFT28 V2.0 and V3.0 4gb vs 8gb

kisslorand, 2024-05-17:
> How is that even possible to store nearly 8gb on a 4gb.

It's not possible.

> Im surprised its running so well.

Me too.

> I guess that's the difference between the MKS TFT28 V2.0 and V3.0 4gb vs 8gb

Could be, among other stuff.

closed 2024-11-19 by kisslorand