kisslorand/BTT-TFT-FW #18: TFT Artillery Genius Pro

nagyrobi, 2024-11-18:
I have an Artillery Genius Pro with a TFT based on STM23F107VC. I was able to replace its bootloader with a chirp version, much better.

However the TFT binary itself doesn't run.
- with `MKS_TFT28_V4.0.27.x.bin` it doesn't reboot after update finishes. pressing reset manually results in "Booting..." message and nothing happens
- with `MKS_TFT28_V3.0.27.x.bin` I get a white screen after rebooting
- with `MKS_TFT28_NEW_GENIUS.27.x.bin` I get a screen with a few blank lines like below:

[image]

What else should I try?

kisslorand, 2024-11-18:
Did you also provide the rest of the files needed for the FW (config.ini, fonts, icons)?

nagyrobi, 2024-11-18:
Sure!

kisslorand, 2024-11-18:
As an ultimate solution, you might get an X1 or X2 TFT and use that in your Genius Pro or even better, get a bigger BTT screen.

Later edit:
Ah! I just remembered, when switching from original Artillery firmware to BTT based FW there are no fonts to display any text. It could be you case too, the update is going as planned but there are no fonts to display any text. Just try to update again and wait a few minutes for the update to finish.

kisslorand, 2025-08-26:
@nagyrobi 
Any update on this issue?

nagyrobi, 2025-08-26:
Gave back the printer and got a Creality instead.

closed 2025-08-26 by nagyrobi