kisslorand/BTT-TFT-FW #19: Compatibility issue with RRF 3.5.4

KirkBigelow, 2025-01-05:
After debugging with RRF 3.5.4 with a BigTreeTech SKR2 controller and the TFT50 V3.0.1.  The only Gcode command that work are M105 and M92 during startup and screen reset and the Estop M112.  Everything else is dead that I have tested. No terminal. No move or heater gcode are seen on the SKR2 debug side.  The screen does not update after startup/restart with controller data.   I did not test every possible button.  The interface in use is the serial TFT interface.   Does not seem compatible with RRF firmware.  Oh, tried to print a file and nothing there too. 

Yes, the RRF config.ini file was used for the TFT firmware load.  The file name changed to config.ini.CUR.

Same result with both this firmware and BigTreeTech firmware.

I know this is out of scope but nice hardware, but the firmware is lacking.  Would be nice to use the knob to move or set temperatures.  I know this is out of scope.  If I had the magic, I would remove the Up and Down buttons and replace with the knob or do both.  Everything else would stay the same.  Thanks.

effgarces, 2025-01-20:
I'm also having issues with RRF support, while in touch mode i can't control, move or do anything with the printer, but as soon as I change to marlin mode it suddendly "receives" all the instructions that I tried sending before

kisslorand, 2025-01-21:
I got the time to check RRF part, I found some issues. I do not have a RRF printer, I am working just on a TFT and an SKR2 MB.
Can you guys tell me what TFTs do you have so I can upload test FW for you to check?

effgarces, 2025-01-21:
I have two TFT's the TFT35 (i use this file to flash it BIGTREE_TFT35_V3.0_E3.27.x.bin) and the TFT24 (BIGTREE_TFT24_V1.1.27.x.bin)

Thanks for taking a look at this.

kisslorand, 2025-01-21:
@effgarces 
Here are the FWs to test on your TFTs.:
[BIGTREE_TFT35_V3.0.27.x.zip](https://github.com/user-attachments/files/18490388/BIGTREE_TFT35_V3.0.27.x.zip)
[BIGTREE_TFT24_V1.1.27.x.zip](https://github.com/user-attachments/files/18490410/BIGTREE_TFT24_V1.1.27.x.zip)

kisslorand, 2025-01-21:
@effgarces 

Here are the FWs for your TFTs:
[BIGTREE_TFT24_V1.1.27.x.zip](https://github.com/user-attachments/files/18490782/BIGTREE_TFT24_V1.1.27.x.zip)
[BIGTREE_TFT35_V3.0.27.x.zip](https://github.com/user-attachments/files/18490809/BIGTREE_TFT35_V3.0.27.x.zip)

BTW, the issue you reported (#22) has been fixed.

effgarces, 2025-01-21:
As soon as I get home I will give feedback on this.

Yes, I've seen and flashed the newest version, it's working perfectly thank you, hadn't gotten around to answer the the issue, srry.

effgarces, 2025-01-21:
I just tested on both printers and the problem persists. I did noticed that when issuing, for instance a "Home All" command, if I then changed to Marlin mode it would execute that command, the behaviour happens on both TFT's

kisslorand, 2025-01-21:
Please excuse my ignorance as I never used RRF, how do you issue a "Home All" command?

kisslorand, 2025-01-21:
@effgarces 
Check these FWs:

[BIGTREE_TFT35_V3.0.27.x.zip](https://github.com/user-attachments/files/18496543/BIGTREE_TFT35_V3.0.27.x.zip)
[BIGTREE_TFT24_V1.1.27.x.zip](https://github.com/user-attachments/files/18496558/BIGTREE_TFT24_V1.1.27.x.zip)

On my setup with a HW communication monitor (serial port sniffer) I do not see any issue homing, moving X, Y or Z, temperature setting, macro selecting and loading, fan, speed flow control, even the BLTouch functions are working. Terminal also working just fine.

effgarces, 2025-01-21:
> Please excuse my ignorance as I never used RRF, how do you issue a "Home All" command from the TFT?
> 
> Later edit: I think I got it... Working on it.

Sorry, just got back to the printer. The home all command is in "Menu" > "Movement" > "Home all" > "Home all"

I also flashed the new FW, but the problem persists,any function that should cause any kind of "movement" of the printer does not do anything, but as soon as I change to marlin mode they all get executed, even the terminal, I send commands via the tft but they never "arrive" but changing the mode triggers all of them. I did however noticed something, the macro selecting, does not show up on the screen and the info screen says that the system is "unknown".

To rule out any settings that might causing issues, I reflashed everything, icons, language pack, and config_rrf.ini and left them exactly as I downloaded them, no change.

effgarces, 2025-01-21:
Firmware version 3.5.4, that triggered a memory,  I've recently upgraded to version 3.6 beta 2, due to a better input shaping implementation, but,I'm guessing, something in it is causing this problem. I will try to downgrade one of the printers tomorrow and get back to you.

kisslorand, 2025-01-21:
FW 3.6.0 Beta breaks the communication protocol. It doesn't respond with "ok" acknowledgement to the TFT. 

This is my log between the TFT and SKR2 motherboard for 3.6.0 beta3:
```
TFT         >> M105
Motherboard >> T:108.4 /0.0 T0:108.4 /0.0 B:22.3 /0.0
TFT         >> M92
Motherboard >> Steps/mm: X: 80.000, Y: 80.000, Z: 400.000, E: 93.000
```

and this one is for 3.5.4:
```
TFT         >> M105
Motherboard >> ok T:108.4 /0.0 T0:108.4 /0.0 B:22.3 /0.0
TFT         >> M92
Motherboard >> Steps/mm: X: 80.000, Y: 80.000, Z: 400.000, E: 93.000
Motherboard >> ok
TFT         >> M115
Motherboard >> FIRMWARE_NAME: RepRapFirmware for STM32F4 based Boards FIRMWARE_VERSION: 3.5.4 ELECTRONICS: STM32F4 FIRMWARE_DATE: 2024-11-27 23:16:40
Motherboard >> ok
TFT         >> N0 M552*33
Motherboard >> WiFi module is disabled
Motherboard >> ok
```

You can see how with 3.5.4 the motherboard responds with "ok" to each commend/request from the TFT.

If RepRap doesn't change it back to respond with "ok" I am afraid it will be a hard case to make this TFT FW compatible with it again.

effgarces, 2025-01-22:
Awesome, as soon as I get home I will test these and give feedback

effgarces, 2025-01-22:
Flashed both TFT's and tested them, all functions seem to be working ok, I can home, list macros, move, etc. I also updated RRF to the latest beta 3, and retested, all seems to be in working order.

Thank you @kisslorand and @jaysuk for saving these TFT's from the e-waste.

effgarces, 2025-01-23:
Yes, I will be doing a bit of 3d printing on the next few days to calibrate the printers and while at it I will also test the TFT's. So far they have been behaving well

KirkBigelow, 2025-02-03:
I loaded the latest I could find and it did not work on a TFT50.

kisslorand, 2025-02-03:
That's most likely because the repo is not updated with the RRF fix, I am waiting for more feedback. Let me know what's the exact TFT you have and I'll compile a FW for you for testing.

effgarces, 2025-02-03:
Hi,
I've been printing on and off during this last week, especially with my delta that has the TFT28 and the TFT has been holding up and working well, and so far I haven't noticed nothing showstopping. I did however noticed that on the discord, that the commit https://github.com/Duet3D/RepRapFirmware/commit/0623211dccfa634dd7ae0075beaccb0dc415a124 , might cause issues, but after looking through the code I did confirm that the TFT uses M408 S0, so it doesn't the affect us right now.

MenkeVincent, 2025-02-08:
> [@effgarces](https://github.com/effgarces) Check these FWs:
>
> [BIGTREE_TFT35_V3.0.27.x.zip](https://github.com/user-attachments/files/18496543/BIGTREE_TFT35_V3.0.27.x.zip) [...]

Thanks for the Great Work  was sent her by one of the Team Gloomy guys and now everything works for me

KirkBigelow, 2025-02-10:
Kiss Lorand,

    The TFT is a TFT50 V3.0.1 with a F205 processor.

Thanks,

Kirk

kisslorand, 2025-02-10:
@KirkBigelow
Here's the FW for your TFT:
[BIGTREE_GD_TFT50_V3.0.27.x.zip](https://github.com/user-attachments/files/18738257/BIGTREE_GD_TFT50_V3.0.27.x.zip)

It is a release candidate version, includes bugfixes, stability and speed enhances. If I do not find any hiccups in the next few days, I will release it.

kisslorand, 2025-02-13:
The repo is updated with the RRF compatibility fix and others (M118 fix for RRF). Go grab your version of FW!

KirkBigelow, 2025-02-15:
Hmm,  I loaded this and the new files from repo and the results were the same.  I assume the GD means it is for a gigadevice MPU.  My TFT does have this MPU.   I can change the Bed temp sensor input and have to reset the device to see the value change.  I tested this with SKR 2 and Octopus Pro  boards.

Kirk

kisslorand, 2025-02-16:
Can you show me a log?

KirkBigelow, 2025-02-23:
Here is the log from the SKR 2 running RRF 3.5.4.

You will see the display request the temperatures and set the emulation mode to merlin.

Each Aux output occurs only after display reset.

It does pick up the correct temperature value as seen via the USB interface.

It seems the max negative value on the display is -99 versus the sent -273.1 value.   I assume the display maxs out at -99.

>>> M111 P3 S1
SENDING:M111 P3 S1
Debugging enabled for modules: Gcodes(3 - 0xffffffff)
Debugging disabled for modules: Platform(0) Network(1) Webserver(2) Move(4) Heat(5) Kinematics(6) InputShaping(7) Debug(8) PrintMonitor(9) Storage(10) PortControl(11) DuetExpansion(12) FilamentSensors(13) WiFi(14) Display(15) SbcInterface(16) CAN(17) Expansion(18) Driver(19)
Aux: M105
Aux: M555 P2
USB: M105
ok T0:-273.1 /0.0 B:175.5 /0.0
USB: M105
ok T0:-273.1 /0.0 B:175.5 /0.0
USB: M105
ok T0:-273.1 /0.0 B:175.5 /0.0
Aux: M105
Aux: M555 P2
USB: M105
ok T0:-273.1 /0.0 B:175.5 /0.0
Aux: M105
Aux: M555 P2
USB: M105
ok T0:-273.1 /0.0 B:-273.1 /0.0
USB: M105
ok T0:-273.1 /0.0 B:-273.1 /0.0
USB: M105
ok T0:-273.1 /0.0 B:-273.1 /0.0
USB: M105
ok T0:-273.1 /0.0 B:-273.1 /0.0

Regards,

Kirk

kisslorand, 2025-02-23:
Your SKR2 doesn't respond to "Aux", only to "USB", here's your issue.

KirkBigelow, 2025-02-24:
Kiss Lorand,

    The debug data is taken from pronterface terminal.  Which is showing the response from the USB M105 command which pronterface sends to the SKR which is being displayed in this terminal data.  The USB: and AUX: are the Gcode commands being received by the SKR.

The display is receiving some data because the display shows a status of "3D printer ready" and the bed temperature updates to the correct temp which is the one I have setup.  After the initial connection status, it does not update the temperature or the display terminal fails to operate.

I did just connect power to the display to show that without the serial line connect it does not show a status of "3D printer ready" which it did not.

I checked the loaded version and it is showing Feb 10 2025.

I am cool if you do not want to pursue this.

Thanks,

Kirk

kisslorand, 2025-02-24:
What do you see on the TFT if you open "Terminal" and hit "Send"? Is there any data traffic?

closed 2025-08-26 by kisslorand