kisslorand/BTT-TFT-FW #20: TFT Artillery Sidewinder X2 (GD32F205 chip) no touch scrren support issue

Houseman74, 2025-01-07:
Hi. 
First of all thank you Kisslorand for the great work.

Trying to upload new Kisslorand FW on TFT Controller v1.4 for Artillery Sidewinder X2 that has a 28 inch TFT with GD32F205 chipset.
Successfully updated bootloader to v3.0.5 with short chirp.
For the FW I tried:
MKS_GD_TFT28_V4_0.27.x.bin successfully updated but then resulted in Boot logo Bigtree and nothing more. as attached:
[image]

Finally uploading the:
MKS_GD_TFT28_V4_0.27.x Sound_Fixed.bin FW was successful with all icons in place but the touchscreen does not work:
[image]

What am I missing?
Do I need to change something into the config.ini file?
Best regards and thanks.
Steve

kisslorand, 2025-01-07:
Reboot TFT with an empty "Reset.txt" file on your SD card.
Did you see the config file, fonts, icons load after the FW update?

Houseman74, 2025-01-07:
Thank You. That fixed my issue. Putting reset.txt enabled the TouchScreen calibration and then Touch Screen was ready.

kisslorand, 2025-01-07:
Do not forget to use the "config.ini" file provided in this repository and adjust it to your needs.

closed 2025-01-13 by kisslorand