kisslorand/BTT-TFT-FW #23: Cannot do ABL

K0UDA, 2025-02-24:
When doing ABL it wont promt save to eeprom. After ABL menu is stuck, it wont open mesh and its stuck till next power reset.

K0UDA, 2025-02-24:
Edit:
It completely freezes communication with printer.

kisslorand, 2025-02-24:
What happens if you issue a `G29` command in the terminal?

K0UDA, 2025-02-24:
It starts ABL, while giving echo:busy: processing.
After it finished it gave me grid numbers...

I also tried to flash ur older firmware, and it seems that it wont read the firmware.bin ...it skips it so im stuck on 1 FW. This happened before, but ur latest release flashed sucesfully. Now its stuck again.

kisslorand, 2025-02-24:
> Now its stuck again.

What is stuck again?

K0UDA, 2025-02-24:
The firmware thats flashed and cant be reflashed....sorry.
After G29 is finished i can do M500 and it works.

K0UDA, 2025-02-24:
If i do ABL from touch menu, after it finishes...it will freeze connection with printer

kisslorand, 2025-02-24:
I just tested ABL from touch menu:

[image]

kisslorand, 2025-02-24:
Let's dive in further. Start an ABL measurement from the touch and switch to Terminal. What is displayed there after ABL finishes?

K0UDA, 2025-02-24:
actually managed to flash ur older FW and it works now. Should i flash the newest again and try to debug it with you?

kisslorand, 2025-02-24:
Strange, I have done nothing to ABL in any of my FW version.
Flash the newest one and let's try to find out what's going on.

K0UDA, 2025-02-24:
I have BTT TFT35 E3 V3 GD version. So im flashing this: BIGTREE_GD_TFT35_V3.0_E3.27.x.bin

K0UDA, 2025-02-24:
Flashed newest and its not working again. But its not due to ABL...it just disconets the printer and i can reconect it....

kisslorand, 2025-02-24:
Did you monitor communication on Terminal?

After pressing the "ABL" button switch to terminal.

K0UDA, 2025-02-24:
it finished, showed the measured ABM grid and M118 ABL complete, than it stopped the comunication.

kisslorand, 2025-02-24:
Can you please provide a screenshot/picture of it?

K0UDA, 2025-02-24:
[image]
[image]

K0UDA, 2025-02-24:
and now printer is disconnected from display

kisslorand, 2025-02-24:
Not disconnected but doesn't send any commands to it. It is because the printer didn't respond to `M118 P0 ABL Completed`.
I have to find out why...

kisslorand, 2025-02-24:
OK, I think I found the culprit. I am on it fixing that.

kisslorand, 2025-02-24:
Issue fixed. Please test it.

[BIGTREE_GD_TFT35_V3.0_E3.27.x.zip](https://github.com/user-attachments/files/18950698/BIGTREE_GD_TFT35_V3.0_E3.27.x.zip)

K0UDA, 2025-02-24:
Now it works perfectly!

kisslorand, 2025-02-24:
Thank you for reporting this issue, I really appreciate it.

closed 2025-02-24 by kisslorand