kisslorand/BTT-TFT-FW #24: Too long extrusion prevented

Mike-Honcho71, 2025-03-26:
I have a SKR MINI E3 v3 and a tft35 and swapped my screen to this firmware due to the BTT firmware causing the printer to freeze mid print and leave everything on Hotend, bed. This firmware seems to cause the same thing. Im stumped.

Pausing or canceling does nothing I can power cycle and resume the print sometimes it works sometimes it dont.

kisslorand, 2025-03-29:
Do you have by any chance the opportunity to swap your motherboard? 
I have a hunch what's going on but I cannot fix it. 
Are you on a very new version of Marlin? As Marlin got newer and newer it became worse and worse with serial communication on STM32 MCU based motherboards.

I recommend SKR1.4T motherboard, it's the best and most stable from the ones I had to deal with.

Mike-Honcho71, 2025-03-31:
Im using the newest v3 mini board, and Im using marlin 2.1 bugfix I complied feb 23rd. I just swapped to octopi and using it via usb to print has zero issues. So it is an issue with screen comms to board.

kisslorand, 2025-04-06:
No. Using USB you avoided the RS232 port issues the Marlin firmware has, hence the hassle free print with OctoPi.

closed 2025-08-08 by kisslorand