kisslorand/BTT-TFT-FW #25: MKS_GD_TFT28_V4_0.27.x Sound_Fixed.bin bed size for Sidewinder X2

AleSan78, 2025-04-06:
Hello, the bed dimensions are for the Genius, could you compile one for the X2, thank you

kisslorand, 2025-04-06:
The bed dimensions are configurable in `config.ini` file, those dimensions are not hard coded into the firmware.
Please check the config file and adjust there the parameters that suit your printer and/or you preferences.

AleSan78, 2025-04-07:
> The bed dimensions are configurable in `config.ini` file, those dimensions are not hard coded into the firmware. Please check the config file and adjust there the parameters that suit your printer and/or you preferences.

Yes, I saw after posting my question, I was using my old config.ini file, everything works now, thanks. However, the source files are not available, I was wondering, if it would be possible to change and put this in the "configuration.h" file: #define PROGRESS_BAR_COLOR 4 // Default: 0, this is to change the color of the progress bar to blue rather than orange.

kisslorand, 2025-04-07:
Sure, I can compile custom FW for you, whatever config changes you want. However you should wait a little, there's an update waiting to be merged with an exclusive new feature (no Marlin printer has such feature) and  some speed increase.

Just out of curiosity, why do you use the "MKS_GD_TFT28_V4_0.27.x Sound_Fixed.bin" firmware file? Do you have sound issues with the regular "MKS_GD_TFT28_V4_0.27.x.bin" firmware?

AleSan78, 2025-04-07:
Yes, I didn't have any sound, just the one when turning on the printer.

kisslorand, 2025-04-08:
Here's a FW compiled for you that includes the new feature also which I hope will be to a great use. Do not forget to edit the included `config.ini` file according to your needs. It has to be updated on the TFT as it contains a new config parameter.

[mkstft28evo.zip](https://github.com/user-attachments/files/19645554/mkstft28evo.zip)

AleSan78, 2025-04-09:
Thank you very much, you are very kind, I will try right away.

AleSan78, 2025-04-09:
It doesn't work, I only have white lines and a black background.

kisslorand, 2025-04-09:
Please try also the files in the repository as I updated them with the latest changes. Also a picture would be welcome.
Let me know please.

AleSan78, 2025-04-09:
It's good, it works now, I tried with another SD card, thanks

AleSan78, 2025-04-12:
Hi, a quick question why when I want to flash the firmware I compiled with the latest BTT sources for the TFT GD, the screen flashing stops and remains stuck on "complete".

kisslorand, 2025-04-12:
You're asking this question in a wrong place. I stopped using that FW more than 2 years ago, even before they added the support for GD MCUs so I really have no clue what's going on with BTT FW.

AleSan78, 2025-04-12:
ok, it's fine thanks

closed 2025-05-25 by kisslorand